erdilS/Port-Windows-11-Xiaomi-Pad-5 #17: can't connect to wifi

Edward-Lucas, 2023-01-29:
I followed the guide and installed normally, but there are two big problems.Wifi doesn't work. If you turn off the screen, the device powers off, and if you connect the charger in this state, it enters fastboot mode.

warpreality, 2023-01-29:
wifi fixed after reboot

halal-beef, 2023-01-30:
second issue already known and most likely cant be fixed without qcom gpu source

IliyaBrook, 2023-02-07:
In my case rebooting not make effect =((, i also tried to use bypass and update the driver manually after removed the driver from system management....also no wifi...

priteshz, 2023-04-03:
> [IliyaBrook](/IliyaBrook)

I had similar issue, wifi conn detected and connected at time of setup but if i click next, it shows that it disconnected
I went around the issue by clicking retry<>Next button rapidly

closed 2023-07-09 by erdilS