erdilS/Port-Windows-11-Xiaomi-Pad-5 #25: fsa4480.sys error on startup SYSTEM_SERVICE_EXCEPTION

AnotherJack260, 2023-02-03:
### Description

Followed the tutorial, on first boot i got an error with this driver.
i tried with older drivers, older driver updater and other windows 11 iso. nothing works

### Reproduction steps

```bash
Followed the tutorial, on first boot i got an error with this driver.
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

erdilS, 2023-02-03:
If you get fsa4480.sys blue screen, please do this.

1. Remove USB drivers from Driver pack - delete components\QC8150\Device\DEVICE.SOC_QC8150.NABU\Drivers\USB folder.
2. Reinstall drivers via driverupdater.exe - How to update drivers (https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/blob/main/guide/English/update-en.md)
3. After setup reinstall drivers with USB drivers.

hyunno0, 2023-10-25:
I remove USB drivers from Driver pack - delete components\QC8150\Device\DEVICE.SOC_QC8150.NABU\Drivers\USB folder.
And I installed Windows11 into my MIPAD 5 succesfully...
But, After installing fsa4480.inf Windows went to blue screen...
How can i install USB driver?

hyunno0, 2024-03-04:
I'm also same. when reinstalling USB driver via fsa4480.inf, Windows went to blue screen and reboot, not installed USB driver...

n00b69, 2024-03-04:
> when reinstalling USB driver via fsa4480.inf, Windows went to blue screen and reboot, not installed USB driver...

Did you try installing the latest driver package (4 days ago)? This should fix fsa4480.

You need uefi-v3 to use the drivers.

hyunno0, 2024-03-04:
I try installing the latest driver package (4 days ago)... but blue screen...
And What means "You need uefi-v3 to use the drivers"?

hyunno0, 2024-03-04:
the uefi-v3 supports huaxing and tianma pannel?
My pannel is  Huaxing...

n00b69, 2024-03-04:
> the uefi-v3 supports huaxing and tianma pannel? My pannel is Huaxing...

Yes. The UEFI was updated a few months ago to work on both panels. Huaxing or Tiama UEFI images are no longer needed unless you are using very old drivers.

hyunno0, 2024-03-04:
Thanks 
And I must use only Windows Build: 22631.2861?
I have Windows Build: 22621 ISO file...

hyunno0, 2024-03-05:
Error installing fsa4480 driver: Publisher in signature catalog not trusted...

n00b69, 2024-03-05:
> Error installing fsa4480 driver: Publisher in signature catalog not trusted...

Reinstall Windows, you are updating from very old drivers which does not work

hyunno0, 2024-03-05:
Today, i Reinstalled Windows with 23h2 ...
And i updated new drivers but Error installing fsa4480 driver?...

n00b69, 2024-03-05:
> Today, i Reinstalled Windows with 23h2 ... And i updated new drivers but Error installing fsa4480 driver?...

Can you join the Telegram at https://t.me/nabuwoa

It will be easier to help there

Misha803, 2024-03-07:
Can we close this issue ?

hyunno0, 2024-03-07:
Is the charging issue associated with fsa4480 a perennial task...

hyunno0, 2024-03-09:
When installing fsa4480 driver, Publisher in signature catalog not trusted too...

hyunno0, 2024-03-10:
Error installing fsa4480 driver uploaded 17hr ago too : Publisher in signature catalog not trusted...

n00b69, 2024-03-10:
> Error installing fsa4480 driver uploaded 17hr ago too : Publisher in signature catalog not trusted...

Have you tried [disabling secureboot](https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/blob/main/guide/English/disable-secureboot-en.md)?

hyunno0, 2024-03-10:
In Disabled secureboot, success in installing fsa4480 driver....
But USB HUB(keyboard&mouse) dosn't operate...
Also, when rebooting Windows, Bluescreen...

hyunno0, 2024-03-15:
When this issue will finish?

n00b69, 2024-03-15:
Gustave reworks fsa4480, so maybe new driver pack release will be at March 20-22

For now fix of fsa4480 issues is removing fsa4480 line in nabu.xml and reinstalling drivers.
https://github.com/map220v/MiPad5-Drivers/blob/512b6cb849d592d704baef652580d11c1cd6734b/definitions/Desktop/ARM64/Internal/nabu.xml#L29

n00b69, 2024-03-15:
Extract [this archive](https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/releases/download/1.0/fsa4480.7z) to components\QC8150\Device\DEVICE.SOC_QC8150.NABU\Drivers\USB and reinstall drivers.

n00b69, 2024-03-18:
Try updating this driver manually in device manager

hyunno0, 2024-03-18:
I updated this driver manually in device manager. 
But USB HUB(keyboard&mouse) dosn't operate...
Also, when rebooting Windows was Bluescreen...
Now, the fsa4480 is instelled as a ON Senmiconductor USB Type-C... on device manager...
What fuction is ON Senmiconductor USB Type-C...?

hyunno0, 2024-03-18:
FSA4480 is a high performance USB Type-C™ port multimedia switch which supports analog audio headsets. FSA4480 allows the sharing of a common USB Type-C™ port to pass USB2.0 signal, analog audio, sideband use wires and analog microphone signal. FSA4480 also supports high voltage on SBU port and USB port on USB Type-C™ receptacle side.

hyunno0, 2024-03-31:
After installing new drivers, Bluescreen forever...
Maybe, fsa4480 drivers problem...

hyunno0, 2024-03-31:
And When re-installing all drivers(except fsa4480 drivers), SLM_Child00 on Device Maneger is Error...

n00b69, 2024-05-13:
Can you try latest drivers?

hyunno0, 2024-05-14:
Yes, I did but Bluescreen also...

hyunno0, 2024-05-15:
Yes, I did, but BlueScreen also...

hyunno0, 2024-05-26:
How can I download this TESTnabu-drivers.zip?...

n00b69, 2024-05-26:
USB is in the OEM and Platform folders. Just run OfflineUpdater.cmd with this pack.

hyunno0, 2024-06-05:
The fsa4480 drivers sent me from you is installing failed...

n00b69, 2024-06-05:
What does it say?

hyunno0, 2024-06-07:
Installing the fsa4480 driver on Windows failed...
Do I have to run OfflineUpdater.cmd to reinstall the entire driver...

n00b69, 2024-06-10:
Try installing it manually first in device manager by selecting the unknown
fsa4480 device

hyunno0, 2024-06-11:
Try installing it manually in device manager by selecting the unknown fsa4480 device, but failed...

n00b69, 2024-06-12:
> Try installing it manually in device manager by selecting the unknown fsa4480 device, but failed...

Can you remind me what exactly the issue was again? You don't have working USB, right?

hyunno0, 2024-06-18:
I re-installed the new drivers on https://github.com/map220v/MiPad5-Drivers...
Which device is unknown to Device Manager?...

n00b69, 2024-07-03:
> I re-installed the new drivers on https://github.com/map220v/MiPad5-Drivers... Which device is unknown to Device Manager?...

Can you share an image of your device manager?

hyunno0, 2024-07-05:
[image]

n00b69, 2024-07-15:
USB works in which version of the drivers?

hyunno0, 2024-07-18:
[image]

hyunno0, 2024-07-18:
When I re-installed the new drivers, The above message appears...
And USB not works...

hyunno0, 2024-07-18:
[image]

Also the same error...

hyunno0, 2024-07-18:
[image]

When I installed the old drivers, There is no error...

n00b69, 2024-07-18:
The error in the drivers does not matter, it is related to app packages.

USB works in old drivers?

hyunno0, 2024-07-19:
Yes, USB works in old drivers but not works in latest drivers...

hyunno0, 2024-07-20:
[CODE_OF_CONDUCT.md](https://github.com/user-attachments/files/16319416/CODE_OF_CONDUCT.md)
[ETW.md](https://github.com/user-attachments/files/16319417/ETW.md)
[LICENSE.md](https://github.com/user-attachments/files/16319418/LICENSE.md)
[NTDBG.md](https://github.com/user-attachments/files/16319419/NTDBG.md)
[README.md](https://github.com/user-attachments/files/16319420/README.md)

Here...

n00b69, 2024-07-20:
Do you have the original zip file by any chance?

hyunno0, 2024-07-21:
Why not USB works in latest drivers in my case...
can you check this issue...
And What is my technical support?

n00b69, 2024-07-21:
Can you try (if possible) a clean Windows install with the drivers I linked? If they work, I can ask the developer of the drivers to look into it

hyunno0, 2024-08-11:
There is a lot of problems in USB...
In usb host mode, not charging by the PD of USB HUB but only can charge by Charging Adapter...
In none usb host mode, USB HUB not be recognized...
We hope these things will be improved as soon as possible...

n00b69, 2024-08-12:
Everyone has that issue and it sadly cannot be fixed.

hyunno0, 2024-08-13:
It cannot be fixed forever? or have possibility next time?

hyunno0, 2024-08-15:
This issue be never fixed forever?...

n00b69, 2024-08-16:
Unlikely

NewbieHerePlsHelp, 2024-08-25:
[image]
Somebody help me I don't know what to do in BSOD in my pad 5, I don't know how to do the code thing🥺

n00b69, 2024-08-26:
Try these drivers

https://github.com/map220v/MiPad5-Drivers/releases/tag/2024.02.29-2

NewbieHerePlsHelp, 2024-08-27:
Thanks

 
 

https://github.com/map220v/MiPad5-Drivers/releases/tag/2024.02.29-2

hyunno0, 2024-09-15:
When the latest Drives will post?...

[3 more replies]

closed 2025-02-28 by n00b69