erdilS/Port-Windows-11-Xiaomi-Pad-5 #42: [BUG] - Powerbutton not working

DedBash, 2023-02-20:
### Description

I have now installed windows 11 (in German) according to instructions and have the problem that the power button does not quite what I would think what this should do
1. the powerbotten itself works under android etc normal
2. under windows the Tabelt can be put into sleep mode but it can no longer be woken up so that you have to restart it completely every time (By holding down the power button until the device reboots)

https://user-images.githubusercontent.com/79027536/220103438-d87aac42-f2c3-486d-9654-0088498dd622.mp4

### Reproduction steps

```bash
1. Press the power button briefly so that the screen turns black
2. Try to wake up the display by pressing briefly again
3. Nothing
```

### Logs

```bash
I do not know where there could be logs for it
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

kutlay1653, 2023-02-20:
Hello,
Thanks for mentioning bug but it is already known and developer(s) work on it. Thanks anyway. You can check project status from main page.

Clousry, 2023-02-20:
already known

closed 2023-02-20 by erdilS