erdilS/Port-Windows-11-Xiaomi-Pad-5 #45: Stuck on BSOD automatic start up repair

hummy2805, 2023-03-13:
### Description

after several reboots. windows 11 on mi pad 5 loops at automatic startup repair BSOD
can't choose option in bluescreen such as advanced mode/trouble shooting mode. bcoz no mouse and keyboard function

### Reproduction steps

```bash
after several reboots. windows 11 on mi pad 5 loops at automatic startup repair BSOD
can't choose option in bluescreen such as advanced mode/trouble shooting mode. bcoz no mouse and keyboard function
```

### Logs

```bash
after several reboots. windows 11 on mi pad 5 loops at automatic startup repair BSOD
can't choose option in bluescreen such as advanced mode/trouble shooting mode. bcoz no mouse and keyboard function
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

itayalush, 2023-06-02:
fuck this shit... i have this problem too

shalabh46, 2023-10-17:
i had same problem. how to recover.

Xyy155, 2023-12-09:
try it enter to fastboot and reroot your device with "fastboot flash boot (your root file location in pc)" and type
"fastboot reboot" now your device on android and reinstall windows

bone6000, 2024-01-10:
anyway to recover without reinstalling windows again? I have some configurations and apps installed, i don't want to loose all my job

n00b69, 2024-02-27:
Open a new issue if this still happens, in the meantime the guide and the install method has been updated extensively.

closed 2024-02-27 by n00b69