erdilS/Port-Windows-11-Xiaomi-Pad-5 #48: [QUESTION] - Good news ! Charging work!!! <title>

kmille36, 2023-03-30:
### Question

I can charging while im on windows ( 12.2v 1.3A).

Flash this UEFI: https://github.com/kmille36/TempStorage/blob/main/xiaomi-nabu.img?raw=true

[image]
[image]

You need to plug in charger after windows bootup, otherwise it will not charging...

Also USB type C Apdater work by modify value in Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\ACPI\QCOM0597\0\Device Parameters ( 3 to 1). Then restart windows!

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

priteshz, 2023-04-02:
I am getting charging but the charging percentages keep dropping while being plugged anyway, just like before i used your steps

priteshz, 2023-04-03:
There appears to be a bug with charging percentages (MI Pad is 2 months old and battery is fine)
The percentage reaches 100 and then bounce back to 75, then start increasing again, and then repeats

priteshz, 2023-04-13:
@kmille36 charging stop at 80%, is that intentional?

krvrian, 2023-05-23:
hey i tried this yet it still not charging, any clue?

kmille36, 2023-05-23:
> hey i tried this yet it still not charging, any clue?

Use type c to c charger

krvrian, 2023-05-24:
> Use type c to c charger

oh I tried using macbook's charger and it's charging now, thank you

Emilia-sama, 2023-06-19:
Using the uefi and the newest driver bring BSOD problem, error on code INTERNAL_POWER_ERROR
From my observation, it load windows with 6w charge rate and when it almost done loading it crashed. Same goes to everytime shutting off the pad5 even tho no charger was connected. Switching the registry file doesn't bring any changes.

On event viewer also didnt mentioned the driver that causes this, only notes that it says is the device lose power

n00b69, 2024-02-27:
Charging is enabled by default now due to the USB host mode tweak

closed 2024-02-27 by n00b69