erdilS/Port-Windows-11-Xiaomi-Pad-5 #61: [BUG] - Can't enable auto brightness

Lilobast, 2023-04-15:
### Description

I can't find the auto brightness option
If I'm right, the tablet should have the light sensor working in windows

### Reproduction steps

```bash
Normal install via tutoriel
```

### Logs

```bash
N/A
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

JeromTWinL, 2023-07-16:
According to [status.md](https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/blob/main/guide/English/status.md), Light sensor isn't working

Lilobast, 2023-07-16:
It was modified earlier this month to uncheck the light sensor,
so it is confirm to not work apparently

Lilobast, 2024-01-15:
Fixed with new drivers

closed 2024-01-15 by Lilobast