erdilS/Port-Windows-11-Xiaomi-Pad-5 #64: [BUG] - Tablet boots to fastboot mode randomly, need to flash nabu-boot.img again for booting to windows again

priteshz, 2023-04-19:
### Description

Tablet randomly boots to fastboot, after shutting down or restarting. Need to flash UEFI again in order to boot to windows again.

### Reproduction steps

```bash
1. Install using normal steps with latest drivers
2. use for a while and observe when booting, it sometimes boots to fastboot instead of windows
```

### Logs

```bash
Not available, how should i take them?
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

priteshz, 2023-04-23:
I think it got fixed by running `fastboot flash boot` for both `boot_a` and `boot_b` together. Will back after a few days of usage

xXPOLYGONXx, 2023-05-06:
> I think it got fixed by running `fastboot flash boot` for both `boot_a` and `boot_b` together. Will back after a few days of usage
> 
> Edit: i confirm that this fixed issue for me.

I tried the same and it betters the problem, but when hardresetting the tablet (Holding power button for 10 seconds) sometimes this happens again. No big issue since you can reflash the bootloader, but since you cant wake the tablet when the display turns off it happpened multiple times to my tablet.

priteshz, 2023-05-06:
> I tried the same and it betters the problem, but when hardresetting the tablet (Holding power button for 10 seconds) sometimes this happens again. No big issue since you can reflash the bootloader, [...]

Hi, it is most important to turn off all settings that make tablet sleep. It is a known bug that tablet won't wake up if it goes to sleep. Turn off automatic sleep, and turn also turn off sleep by power button from "Change what power buttons do" setting from control panel. You must make sure that tablet never goes to sleep and all options related to sleep must be hidden or disabled. you should also disable showing sleep option in power menu at start. Consider doing this as long as sleep bug is fixed. Thanks.

xXPOLYGONXx, 2023-05-06:
> > > I think it got fixed by running `fastboot flash boot` for both `boot_a` and `boot_b` together. Will back after a few days of usage
> > > Edit: i confirm that this fixed issue for me.
> > 
> > 
> > I tried the same and it betters the problem, but when hardresetting the tablet (Holding power button for 10 seconds) sometimes this happens again. No big issue since you can reflash the bootloader, but since you cant wake the tablet when the display turns off it happpened multiple times to my tablet.
> 
> Hi, it is most important to turn off all settings that make tablet sleep or make the display go off. It is a known bug that tablet won't wake up if it goes to sleep. Turn off automatic sleep, and turn also turn off sleep by power button from "Change what power buttons do" setting from control panel. You must make sure that tablet never goes to sleep and all options related to sleep must be hidden or disabled. you should also disable showing sleep option in power menu at start. Consider doing this as long as sleep bug is fixed. In case you are worried about display, just set a screen saver with black image or anything else. Thanks.

Thanks for your reply, but I am aware of the screen bug and its mittigations. I just wanted to emphasize the importance of this issue, since the screen bug exists and probably more people will face this issue. And it should be noted that just flashing to a and b partition does not prevent this from happening.

kujourin, 2023-05-16:
Yep I have the same problem and it makes me have to need another computer anytime...

kujourin, 2023-05-19:
Now I have a stupid solution to this problem...
Don't shutdown from windows, switch back to android if you need to.

destill, 2023-06-05:
The same is hapening with my tablet. After randomly reboots it will boot into fastboot. Do somebody know how to fix it?

Youriywot, 2023-06-17:
> The same is hapening with my tablet. After randomly reboots it will boot into fastboot. Do somebody know how to fix it?

No, even the script with rebooting into android and replacing c:\boot.img with boot-uefi does not give anything, the script writes that the data exchange is 0kb, that is, it didn’t even flash, it already infuriates, there is no PC to raise the tablet from fastboot, this is clearly  boot-uefi curve

Youriywot, 2023-06-19:
I figured out if someone flies uefi, put twrp in slot A, and uefi in slot B, in fastboot set_active B, now when uefi crashes the tablet will automatically start slot A with twrp, put uefi-boot in slot B and immediately reboot into the system  , do not change slots in twrp, you will immediately get into fastboot!

Youriywot, 2023-06-23:
It doesn’t help, I just get into fastboot using only windows ...

Youriywot, 2023-07-17:
Fix, install boot android 13 slot_a, boot uefi slot_b, set_active b

Jbrimbelibap, 2023-12-07:
> Fix, install boot android 13 slot_a, boot uefi slot_b, set_active b

I don't think that this fixes anything, once the tablet starts going into fastboot mode uppon booting it doesn't change slot on its own.

By running fastboot getvar all, I noticed that it said that the current slot is unbootable when the bug happens.
To fix it, you don't actually need to reflash the recovery : running "fastboot --set-active" is enough

**So the root cause of our problem is this : a bug in the UEFI causes the tablet to set it's current slot as inactive.**

In my experience, I get the bug everytime the tablet's been off for a while, and I plug in a charger to start it, or boot with a usb device plugged in.

n00b69, 2024-02-27:
Open a new issue if this still happens, in the meantime the guide and the install method has been updated extensively.

closed 2024-02-27 by n00b69