erdilS/Port-Windows-11-Xiaomi-Pad-5 #66: Please someone help increasing windows storage without having to re install

ItsnacioCh, 2023-04-28:
### Question

Help, i want to increase the storage on windows and decrease the storage on android, without having to re install.

Is it possible? Need assistance

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

priteshz, 2023-04-29:
I have removed the userdata partition from windows and merged it with WIN partition. I have later used uninstall script and it worked fine. So i can confirm that this method works without bricking the device.

priteshz, 2023-04-29:
[image]

[image]

kmille36, 2023-05-01:
OK Great !

Emilia-sama, 2023-05-01:
> I have removed the userdata partition from windows and merged it with WIN partition. I have later used uninstall script and it worked fine. So i can confirm that this method works without bricking the device.

can you boot back to android? doesn't uninstall script restore original storage to android? leaving no space for windows at all

priteshz, 2023-05-01:
> can you boot back to android? doesn't uninstall script restore original storage to android? leaving no space for windows at all

Yes, what i meant is that we can safely merge userdata partition from windows and increase windows storage

By mentioning uninstall script i meant that the change is reversible without bricking the device

Warning ⚠️: do not touch any other partition, just carefully delete userdata carefully and expand WINNABU partition with that space.

musapnr, 2023-11-28:
> Lol u can not without losing data both. Need to format partition.

mkpart esp fat32 10.9GB 11.4GB
mkpart win ntfs 11.4GB 209.8GB
mkpart userdata ext4 209.8GB 254GB

I have a tablet with 256GB. I haven't started the installation yet. If I change these commands like this, will it work? I want the Windows size to be larger

n00b69, 2024-02-27:
> mkpart esp fat32 10.9GB 11.4GB mkpart win ntfs 11.4GB 209.8GB mkpart userdata ext4 209.8GB 254GB

Please use the new method, "adb shell partition 198.4" instead

closed 2024-02-27 by n00b69