erdilS/Port-Windows-11-Xiaomi-Pad-5 #70: [BUG] - Windows 11 is slow

priteshz, 2023-05-06:
### Description

The UI feels sluggish

### Reproduction steps

```bash
1. Install with normal steps
2. Observe while using
```

### Logs

```bash
N/A
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

priteshz, 2023-05-06:
[Solution]
1. Turn off transparency
2. Turn off animations
3. From settings > System > About > Advanced system settings > Performance settings > "Adjust for best performance" and click OK (Keep Smooth edges of screen fonts on if you like it)
4. Uninstall default metro apps
5. Reboot and observe that windows is now a lot faster.

Keep habit of using apps optimized for ARM. you will likely continue to use windows and forget android after applying these settings and observing very good performance.

Ilya114, 2023-05-06:
Not a project bug

halal-beef, 2023-05-06:
what Ilya said, this is not a bug

priteshz, 2023-05-06:
Can you try and verify the [Solution] comment and if it gives satisfactory performance improvement, consider adding it to tutorial?

halal-beef, 2023-05-06:
no? this isn't worth adding to the guide this is the fault of windows and not of the mi pad 5

priteshz, 2023-05-06:
understandable, closing

closed 2023-05-06 by priteshz