erdilS/Port-Windows-11-Xiaomi-Pad-5 #82: [BUG] - <Windows reboots into fastboot>

Youriywot, 2023-06-23:
### Description

I use only windows, during a normal shutdown or reboot I get into fastboot, I tried to put TWRP in slot b, at first when uefi crashed, twrp was loaded, it helped for a couple of reboots, flashed with twrp boot-nabu, but then twrp stops loading, only fastboot ..  .

### Reproduction steps

```bash
Reboot
```

### Logs

```bash
Where to get them?
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

Youriywot, 2023-07-17:
Set boot-13android to slot_a, uefi to slot_b, set_active b, now reboots, etc. does not get into fastboot

n00b69, 2024-02-27:
Open a new issue if this still happens, in the meantime the guide and the install method has been updated extensively.

closed 2024-02-27 by n00b69

Hakzel, 2025-04-24:
I also encounter the same problem. Randomly the tablet ends up in Fastboot mode and I need a PC to make it work again. Very frustrating as I want to use the tablet while being on the road where I don't have a PC in my bag all the time :-/

I also tried flashing the recovery images to the boot sectors but it doesn't recover to them