erdilS/Port-Windows-11-Xiaomi-Pad-5 #85: [BUG] - WIFI not quite working on w11 instalation setup

Livb-VRC, 2023-07-08:
### Description

I just finished following the guide and installing w11. Seems to work but when I'm at the windows 11 setup and it asks me to connect to a network, I select one and hit next it says "Oops, you've lost internet connection"

I've tried restarting tablet and connecting to other networks same issue

Tablet is from before 2023

### Reproduction steps

```bash
- Install w11
- Go through setup
- Connect to network and hit next
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

Livb-VRC, 2023-07-08:
For anyone having the same issue: I was able to get around this by:

In this screen, open the icon accessibility and enable on screen keyboard After that Shift + fn + f10 The cmd Will appear Type : OOBE\BYPASSNRO"

closed 2023-07-09 by erdilS

Shad0w0ne, 2023-10-24:
hello, great that help a lot