erdilS/Port-Windows-11-Xiaomi-Pad-5 #87: [BUG] - Standby doesnt work

gianlucadest, 2023-07-10:
### Description

I updated my Xiaomi Pad 5 to the latest drivers and wanted to try out the standby function that got fixed. When i press the power button it just shows some blurred blue horizontal lines in 50% of the display. I've taken a look at the project status which shows that it is possible to do so. Do i need to reinstall everything or do i miss something?

sincerely

Gianluca

### Reproduction steps

```bash
1.) Update to latest drivers
2.) Use Power button to go into standby or rightclick the windows logo->standby
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

halal-beef, 2023-07-13:
There is a tianma uefi you can try here:

https://t.me/nabuwoa/41419

This should fix the issue for you, try and see

gianlucadest, 2023-07-14:
It solved the issue! Thanks!

closed 2023-07-14 by gianlucadest