erdilS/Port-Windows-11-Xiaomi-Pad-5 #100: [QUESTION] - External monitor

Livb-VRC, 2023-08-06:
### Question

I have a usb hub where I connect monitor/mouse/keyboard/mic/camera. The output is usb c 

This works fine for my mac and phone, and on windows 11 mi pad 5 it detects keyboard/mouse but not the monitor.

Is it possible for the monitor to somehow be detected?

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

mohankrishnarao, 2023-08-16:
actually Xiaomi pad 5 type c  port doesn't support video out 
it's not software problem it's hardware problem
so there wont be any fix that issue 
and Xiaomi pad 6 is upgraded to port that allows video out

n00b69, 2024-02-27:
Only possible with DisplayLink certified adapters

closed 2024-02-27 by n00b69