erdilS/Port-Windows-11-Xiaomi-Pad-5 #102: [QUESTION] - Can you update the repo?

palexdev, 2023-08-07:
### Question

There are no instructions on how to update the UEFI
The release here is not the latest...
Can you update and add the guide for updating the UEFI?

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

Misha803, 2024-01-18:
guide already updated

Misha803, 2024-01-18:
@palexdev Guide aready updatted, if you want suggest anything, write here abut it

palexdev, 2024-01-22:
@Misha803 seems all good to me. I mean, I still don't have secure boot, and I still use the .bat file for dual boot, to be honest the Telegram group is quite hard to follow. Like, right now I have 2005 unread messages, and I'm sure that 90% of them are people that failed to follow the guide and are requesting support. That would be an advice, maybe there should be a section for news and a section for support.
But then again, that's another world, and it's on me, so..... thanks for updating the guide :)

closed 2024-01-22 by palexdev