erdilS/Port-Windows-11-Xiaomi-Pad-5 #105: [BUG] - Huaxing screen not working after enabling OTG

senayuki, 2023-08-18:
### Description

The left side (vertical) displaied some strange, and the right side is completely black. I have to reboot the device.

After I restored the registry entries and keep the boot provided by troubleshooting, it returned to normal.

### Reproduction steps

```bash
When I follow the instructions in troubleshooting, flashed boot and editted the registry, OTG is available. But if I lock the screen and turn it on again, the display of the screen will be abnormal.
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

bone6000, 2023-08-26:
same problem to me. when switch off screen strange lines appears in half part of the screen and the only option to recover from that is press the power button 15 secs. huaxing screen also

TUEJJ, 2023-09-02:
I have the same problem

Misha803, 2024-01-18:
this is already solved, update to latest  drivers and UEFI

Misha803, 2024-01-18:
@senayuki this is already solved, update to latest drivers and UEFI

senayuki, 2024-02-02:
> @senayuki this is already solved, update to latest drivers and UEFI

You're right, I upgraded UEFI and driver, and now it's working. Thanks!

closed 2024-02-02 by senayuki