erdilS/Port-Windows-11-Xiaomi-Pad-5 #106: [BUG] - Random Freezes from new driver update

kitsuki99m, 2023-08-20:
### Description

Screen freezes randomly on pad 5. 

1. Before it freezes, lag occurs slowly.
2. Then freezes instantly after a while.

Windows 11 22H2 version 22621.2134

Note: I do not any have log files, no bluescreens occurred.

### Reproduction steps

```bash
Installed the newer version of the OS, followed the instructions in installing the latest drivers to a fresh installed OS. After browsing for a while, screen lags and few minutes later it freezes. Using the Pad 5 while charging.
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

kujourin, 2023-10-15:
Automatic screen rotation cannot be used after updating the driver

kujourin, 2023-10-15:
after updating the driver... GPU cannot used....

Misha803, 2024-01-17:
update to latest drivers and tell me if you solve this

kujourin, 2024-01-18:
@Misha803 solved

closed 2024-02-25 by n00b69