erdilS/Port-Windows-11-Xiaomi-Pad-5 #122: Audio distortion [BUG] - <title>

Ppkaim41, 2023-10-27:
### Description

Sound distortion problem while using Bluetooth,using gom player, cyberlink powerdvd ultra,vlc, windows media player,same problem with all software..no codecs available to change as aac,ldac..no dolby atmos driver....please fix the issue.. Bluetooth works well with pc and mac,can run kbps upto 1000..

### Reproduction steps

```bash
Audio driver update required as dolby atmos,or similar
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

n00b69, 2024-02-27:
Does this still happen on the latest drivers?

Pratikji44, 2024-03-09:
> Does this still happen on the latest drivers?

No issues now ..but can't change codecs

n00b69, 2024-03-10:
Closing because it's resolved, for codecs you can try contacting map220v or gus33000

closed 2024-03-10 by n00b69