erdilS/Port-Windows-11-Xiaomi-Pad-5 #123: [BUG] - Dead tablet

itayalush, 2023-10-29:
### Description

Fastboot not working, I accidentally destroyed Android, Battery doesn't even charge and just drains, I can't get it working.

### Reproduction steps

```bash
It doesn't boot to fastboot, the battery drains in less than a minute.
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

itayalush, 2023-10-29:
@erdilS pls help

Xyy155, 2023-12-08:
uefi its use battery too much

erictoman, 2023-12-19:
I can help

Misha803, 2024-01-18:
@itayalush go to service centre or ask person ho have auth account, you need flash it in edl

closed 2024-02-27 by n00b69

itayalush, 2024-04-01:
@Misha803 I got an EDL cable but it doesn't do anything.
@erictoman How can I DM on GitHub? you can email me in itay.alush.02@gmail.com

If anyone was wondering, I do have TWRP but it's black screen with only a little line at the top where display works. and it doesn't connect with ADB or MTP

n00b69, 2024-04-01:
> @Misha803 I got an EDL cable but it doesn't do anything.

You can use https://mrauthtool.com/ to EDL flash your nabu (or other Xiaomi device). It will cost around 3-5 dollars.

Credits which are needed to use the flashing tool can be purchased with USDT through the Telegram vendors which are listed on the homepage.

You can use Telegram's @wallet to purchase USDT easily, they do however have higher fees than regular exchanges.

erictoman, 2024-04-15:
> @Misha803 I got an EDL cable but it doesn't do anything.
> 
> @erictoman How can I DM on GitHub? you can email me `itay.alush.02@gmail.com` I'm only 11 but I do know a lot of things like how to install custom ROMs, Install Linux, and I even know Hackintosh (Porting macOS on normal PC) That's why I can't pay for an EDL authorized account. And also, how do I use an EDL authorized account? I can't enter Android, Can't enter Fastboot, TWRP has black screen and does not connect to ADB or MTP.

Are you still interested?

Fizzi1470, 2025-10-28:
> Are you still interested?

I am if your offer is still available