erdilS/Port-Windows-11-Xiaomi-Pad-5 #124: Stuck on windows auto repair blue screen

Code691, 2023-10-29:
### Description

Everything worked great until my windows restarted for auto-repair, the windows auto repair blue screen came up AND i couldnt connect to any input method and the touch screen is not responsive to click on shut down. and when i tried to hard shut it down through the physical power button it restarted with the message that windows did not start properly, and again is asking me through the blue screen to click on restart, again i am unable to click that and i am stuck in the blue screen.

### Reproduction steps

```bash
Need to restart windows to reload properly through windows OS restart
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

enggariel, 2023-11-20:
i also experiencing the same problem. how to solve it ?

warriorkidz, 2024-01-05:
try this https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/issues/125#issuecomment-1879041337

n00b69, 2024-02-27:
Reinstall windows

closed 2024-02-27 by n00b69