erdilS/Port-Windows-11-Xiaomi-Pad-5 #125: Any Code or method to restart windows when it starts in auto repair blue screen?

Code691, 2023-10-29:
### Question

when i start my tablet on windows it shows blue screen for auto repair and asks me to click on restart while i cannot have any mouse/keyboard connected nor the touch method is responsive so i am stuck. any code or help on the matter will be appreciated. thanks

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

warriorkidz, 2024-01-05:
I've fixed my self by combining some tutorial from different pages.
so do it carefully and not missing step.

fastboot boot recovery.img

adb push msc.sh /sbin/

adb shell sh /sbin/msc.sh

Boot the recovery image with fastboot boot <recovery.img>

Run the mass storage script with adb shell

Open cmd as admin

Type diskpart to start diskpart (shocker)

Find the ESP Volume with list volume, i should be the one named ESPNABU

Select the ESP volume with select volume <number>

Assign the letter with assign letter=y

Exit diskpart with exit

Run this command to re enable testmode bcdedit /store Y:\EFI\Microsoft\BOOT\BCD /set {default} testsigning on

Reboot the tablet into bootloader and boot your UEFI image and Windows should boot

Misha803, 2024-01-31:
@warriorkidz bro, it's very old instructions, please delete it, now no need use test mode, just update to latest drivers and uefi

Misha803, 2024-01-31:
@Code691 update to latest drivers and UEFI, no need to use test mode now, if you already solve it, please close this issue

closed 2024-02-27 by n00b69