erdilS/Port-Windows-11-Xiaomi-Pad-5 #126: Partition issue...

hyunno0, 2023-11-09:
### Question

One issue was happened in making the partition bellow...
Let me know a solution...

------------------------------------------------------------------------------------
(parted) mkpart esp fat32 13.6GB 14.1GB
sh: udevadm: inaccessible or not found
Error: Partition(s) 15 on /dev/block/sda have been written, but we have been
unable to inform the kernel of the change, probably because it/they are in use.
As a result, the old partition(s) will remain in use.  You should reboot now
before making further changes.
Ignore/Cancel? i
sh: udevadm: inaccessible or not found

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

Misha803, 2024-01-17:
Bro, use new easy guide on github instead this, or contact me in telegram - https://t.me/misha_803 and i help you install

n00b69, 2024-02-25:
Closing because this issue is due to the old guide

closed 2024-02-25 by n00b69