erdilS/Port-Windows-11-Xiaomi-Pad-5 #127: There is a big problem in battery state on Windows 11

hyunno0, 2023-11-19:
### Question

The display of remaining battery power does not change as the battery is discharged...
I think it's a battery-related drivers problem...
How can I fit it?...

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

Jbrimbelibap, 2023-12-06:
Indeed, it appears to not be accurate at all, whether when charging or discharging, even the displayed value changes a lot after a reboot.

n00b69, 2024-02-27:
The battery % only updates after it has dropped 30% whilst in Windows. Example: if you boot with 90% battery, it will show an accurate % once it gets to 60%.

You can try using ReviOS, as users have reported the battery working fine there.

Closing because this is unfixable.