erdilS/Port-Windows-11-Xiaomi-Pad-5 #141: [BUG] - standby issue

mich-de, 2024-01-07:
### Description

hello
Sometimes the standby mode doesn't work properly. When I reactivate the screen using the power button, there's either a small white line or the panel has maximum brightness. I force a shutdown, but upon restart, it goes into Fastboot mode. When I try to start the recovery from my PC, I get this error:
'fastboot boot recovery.img Sending 'boot.img' (62256 KB) OKAY [ 1.442s] Booting FAILED (remote: 'Failed to load/authenticate boot image: Load Error')'.

using Windows 11 Pro Insider Build 26020.

The solution I have involves these commands: 
fastboot erase misc
fastboot set_active a 
fastboot reboot recovery

### Reproduction steps

```bash
The steps to reproduce the issue are:

    I let the tablet go into standby mode.
    I press the power button briefly.
    It comes back from standby but doesn't display the screen correctly, showing only a single white line, as if the video driver isn't functioning properly
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

Misha803, 2024-01-16:
If you get this error "FAILED (remote: 'Failed to load/authenticate boot image: Load Error')'." just use fastboot set_active other command and then try your command again

Misha803, 2024-01-18:
@mich-de If you get this error "FAILED (remote: 'Failed to load/authenticate boot image: Load Error')'." just use fastboot set_active other command and then try your command again

Misha803, 2024-01-24:
@mich-de if you need any help, contact me in Telegram - https://t.me/misha_803

Misha803, 2024-03-07:
@mich-de are you here? You still have this problem? I will close this issue  soon

mich-de, 2024-03-08:
No solved

mich-de, 2024-03-08:
[image]

n00b69, 2024-03-10:
Does it still go to fastboot mode when you reboot? Or is it fixed when you reboot?

It seems like your nabu has been repaired in the past, this issue is quite rare and only affects some specific panel configurations, and unfortunately seems like it cannot be fixed

smt188, 2024-03-10:
In the past, in 2023, with other drivers, it sometimes went into fastboot mode. Now, after rebooting, it works again until it enters standby mode.

mich-de, 2024-03-10:
Some as smt188

n00b69, 2024-03-10:
Did either of you buy your nabu used/refurbished or did you ever get it repaired?

smt188, 2024-03-18:
In my case is brand new

n00b69, 2024-07-03:
Closing the issue because it is probably not ever going to be fixed, it luckily seems to only affect a very small amount of devices.

Current "workaround" is to either reboot the device, or to press the power button a few times to go back into sleep mode and wake it up again, until it is fixed.

closed 2024-07-03 by n00b69