erdilS/Port-Windows-11-Xiaomi-Pad-5 #149: USB Drivers issue

hyunno0, 2024-02-08:
### Description

If Removing the components\QC8150\Device\DEVICE.SOC_QC8150.NABU\Drivers\USB folder, Drivers not installl...

### Reproduction steps

```bash
If Removing the components\QC8150\Device\DEVICE.SOC_QC8150.NABU\Drivers\USB folder, Drivers not installl...
```

### Logs

```bash
If Removing the components\QC8150\Device\DEVICE.SOC_QC8150.NABU\Drivers\USB folder, Drivers not installl...
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

Misha803, 2024-02-14:
@hyunno0 we update deivers yesterday, try eeinstall it againg (with removed USB folder)

n00b69, 2024-02-27:
Reopen an issue if this still happens on the latest drivers release

closed 2024-02-27 by n00b69