erdilS/Port-Windows-11-Xiaomi-Pad-5 #155: Charging problem...

hyunno0, 2024-02-23:
### Description

It has a big problem with Windows mode that doesn't charge...

### Reproduction steps

```bash
It has a big problem with Windows mode that doesn't charge...
```

### Agreement

- [X] I am using the latest Drivers.
- [X] I am running Windows 11 On The Tablet.
- [X] There is no issue already describing my problem.

Misha803, 2024-02-24:
Charging in Windows only works on specific cables/chargers. One of the requirements appears to be an extra chip in the cable, such as the one on (official) Xiaomi (fast charging) cables.

Even with a working cable and charger, charging is usually rather slow.

IF YOU USE A POWERED USB HUB, DISABLE USB HOST MODE (enabled by default) OR YOU CAN DAMAGE YOUR NABU

Misha803, 2024-02-24:
@hyunno0 
Charging in Windows only works on specific cables/chargers. One of the requirements appears to be an extra chip in the cable, such as the one on (official) Xiaomi (fast charging) cables.

Even with a working cable and charger, charging is usually rather slow.

IF YOU USE A POWERED USB HUB, DISABLE USB HOST MODE (enabled by default) OR YOU CAN DAMAGE YOUR NABU

hyunno0, 2024-02-26:
When We use a powered USB Hub in Android Mode, not Windows Mode,  We must disable USB HOST MODE or  Our device can be damaged ?

hyunno0, 2024-02-26:
How can we  disable USB HOST MODE ?...

hyunno0, 2024-02-26:
C-types USB Hub don't matter, but USB Hub using A-type to C-type USB Gender are dangerous...
Also, there is no A-type USB Hub that supports charging...

Misha803, 2024-02-26:
@hyunno0 just don't use powered USB hub

Misha803, 2024-02-26:
> e

no only in WIndows

n00b69, 2024-02-27:
> How can we disable USB HOST MODE ?...

https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/blob/main/guide/English/troubleshooting-en.md#charging-in-windows-does-not-work

hyunno0, 2024-03-04:
When trying to charge with many types of cables, some cables show charging mode but not charging (current not flowing). This seems to be a problem with Windows drivers...
What is the solution?...

hyunno0, 2024-03-04:
[image]

1 unknown device is the same as the image...
Which driver should I install?...

n00b69, 2024-03-04:
> When trying to charge with many types of cables, some cables show charging mode but not charging (current not flowing). This seems to be a problem with Windows drivers... What is the solution?...

Charging is super slow. 10% battery in maybe 45 minutes

closed 2024-03-15 by n00b69