erdilS/Port-Windows-11-Xiaomi-Pad-5 #158: [BUG] - WoA Helper app

galaxysollector, 2024-02-27:
### Description

### Setup - Android
> [!NOTE]
> If you are unable to move files to the Windows folder, it means you shut down Windows instead of restarting it. To fix this issue, boot back to Windows and use restart, then as it restarts boot to fastboot and use it to return to Android

- Download and install the [WOA Helper app](https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/releases/download/dualboot/woahelper.apk), then open it and grant it root access.
- Download the [UEFI image](https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/releases/download/UEFI/uefi-v3.img) and place it inside the folder named `UEFI` in your internal storage, if this folder does not exist, create it.
- Return to the WOA Helper app and press the `Back up Android boot` button. Select both the `Windows` and `Android` options.
- Press the `Mount Windows` button, then download and move [StA_Installer_nabu.exe](https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/releases/download/dualboot/StA_Installer_nabu.exe) to the newly created `Windows` folder in your internal storage.
- Return to the WOA Helper app and press `Quickboot to Windows`.

In this situation, if I press the `Mount Windows` button, it failed and prints 'something wents wrong. NTFS magisk module recommended'. I cliked 'Yes' in the app and it requires reboot, then I rebooted but no modules were shown in magisk manager. I think that WoA Helper app could not install the module automatically. I extracted the NTFS magisk module from apk and manually installed the module in magisk manager, and finally WoA Helper can mount Windows partition.

### Reproduction steps

```bash
Press the `Mount Windows` button in WoA Helper app and if NTFS module not installed, it will not able to mount Windows partition. I used this rom. https://xdaforums.com/t/rom-sk-aosp-s1-4-1-android-14-nabu-android-14-0-0_r17-22-12-2023.4634820/ I solved this problem by manually installing NTFS module in magisk manager app.
```

### Agreement

- [X] I am using the latest Drivers and UEFI.
- [X] I am running Windows 11 on the tablet.
- [X] There is no issue already describing my problem.

Misha803, 2024-02-27:
@galaxysollector thanks for report, I will contact dev

n00b69, 2024-02-27:
Are you using official Magisk?

galaxysollector, 2024-02-27:
> Are you using official Magisk?

Yes, I'm using official latest 27.0 version.

n00b69, 2024-03-04:
This is fixed in the latest release which HOPEFULLY will be uploaded today

closed 2024-03-04 by n00b69

n00b69, 2024-03-14:
It took some time but it should be fixed now