erdilS/Port-Windows-11-Xiaomi-Pad-5 #160: [BUG] - Flash light does not work.

galaxysollector, 2024-02-28:
### Description

 ## How to use Flashlight 

 - Download [Flashlight.7z](https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/releases/download/1.0/flashlight_fix.7z) and unzip to any folder

> Press any key to disable it

In this situation, I ran flashlight.exe but it does not work.

> [image] to enable flashlight 

### Reproduction steps

```bash
Run flashlight.exe
```

### Agreement

- [X] I am using the latest Drivers and UEFI.
- [X] I am running Windows 11 on the tablet.
- [X] There is no issue already describing my problem.

Misha803, 2024-02-28:
@galaxysollector can you show me screenshot of device manager? it there any devices with errors? you also can try any flash light app from Microsoft store

galaxysollector, 2024-02-29:
> @galaxysollector can you show me screenshot of device manager? it there any devices with errors? you also can try any flash light app from Microsoft store

There is no error in device manager but the result is same.

Misha803, 2024-02-29:
> > @galaxysollector can you show me screenshot of device manager? it there any devices with errors? you also can try any flash light app from Microsoft store
> 
> There is no error in device manager but the result is same.

Did you tried  any flashlight app?

galaxysollector, 2024-03-02:
> > > @galaxysollector can you show me screenshot of device manager? it there any devices with errors? you also can try any flash light app from Microsoft store
> > 
> > 
> > There is no error in device manager but the result is same.
> 
> Did you tried any flashlight app?

Yes but it looks like not working. I updated driver today but the result is still same.

Misha803, 2024-03-04:
hm, strange, if you sure that in device manager you haven't any devices with errors i will report it to dev

Misha803, 2024-03-04:
cab you check cameras page and send me screenshot ?

Misha803, 2024-03-05:
dev say that flashlight was broken in latest drivers, just wait for fix

closed 2024-07-03 by n00b69