erdilS/Port-Windows-11-Xiaomi-Pad-5 #172: [QUESTION] - [POC] Install AtlasOS

DedBash, 2024-03-28:
### Question

Hello, i have tried to install [AtlasOS](https://atlasos.net/) and have to say that it has worked so far and i could not find any problems, but unfortunately i can not test if the keyboard still works because i do not own an official mi keyboard

[image]

[image]

The goal of the POC was to get the ram down a bit and also to take load off the CPU and that worked out well, so currently I can recommend to install AltasOS to get more performance out of the device.

And my question is whether you want to include something like this in the guide (can also write it if desired)

POC = proof of concept

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

n00b69, 2024-03-28:
I don't recommend AtlasOS because you cannot use the shutdown and restart functions (they don't work properly)

DedBash, 2024-03-28:
I can shut down the device and then start it again and even a restart works without any problems

n00b69, 2024-03-29:
> I can shut down the device and then start it again and even a restart works without any problems

So even if you run Windows for an hour, pressing restart actually restarts it? Or do you have to hold power for 8 seconds?

DedBash, 2024-04-04:
sorry for the late reply, but yes no matter how i restart the device restarts without any problems

n00b69, 2024-07-03:
Late answer too but yes AtlasOS is now stable and is in fact recommended as well

closed 2024-07-03 by n00b69