erdilS/Port-Windows-11-Xiaomi-Pad-5 #173: [BUG] - Keyboard not worked on repair mode

KernelSU123, 2024-03-31:
### Description

I'm facing a problem, I need to update a driver, but to do that I need to disable the digital signature of the drivers, to do that I need to reboot the OS while holding down shift, and get into recovery mode, but I can't select "advanced boot options" as my keyboard is disabled in repair mode, fix it in uefi no secureboot and reply me. Thanks

### Reproduction steps

```bash
Enable keyboard in uefi(bios)
```

### Agreement

- [X] I am using the latest Drivers and UEFI.
- [X] I am running Windows 11 on the tablet.
- [X] There is no issue already describing my problem.

n00b69, 2024-06-09:
Not really possible to fix, and what driver are you trying to update? Drivers can be updated with OnlineUpdater in the driver zip.

closed 2024-06-18 by n00b69

KernelSU123, 2024-06-30:
>  what driver are you trying to update?

I'm trying to update the adb drivers