erdilS/Port-Windows-11-Xiaomi-Pad-5 #187: Low audio quality

AdrMC2, 2024-04-28:
### Description

Hello, the sound quality is quite terrible. It is heard with low quality, and since these are the only drivers there are for this tablet with Windows, I wanted to ask if anyone else has the same problem and if there would be any option to improve the audio quality. Otherwise, the Windows tablet is perfect

### Reproduction steps

```bash
Low quality
```

### Agreement

- [X] I am using the latest Drivers and UEFI.
- [X] I am running Windows 11 on the tablet.
- [X] There is no issue already describing my problem.

Jbrimbelibap, 2024-04-30:
Could be an issue specific to your tablet, speakers sound good on mine

n00b69, 2024-04-30:
Asking about it in the Telegram, it seemslike most people have this issue. The audio in Windows is nit as good as in Android.

AdrMC2, 2024-05-02:
Hello, I was doing sound tests on the tablet in Android mode and Windows mode and there is a big difference in the sound. In Android mode you can hear it with very good quality, the sound is very clear, and in Windows mode it completely loses the quality that it has in Android, I looked at the drivers to see if they were updated and everything was fine, but it still sounds bad

n00b69, 2024-07-03:
Likely unfixable, the same issue also affects other Android-WoA devices

closed 2024-07-03 by n00b69