erdilS/Port-Windows-11-Xiaomi-Pad-5 #189: [QUESTION] - Stuck on bootloop MIUI

KevsCoder, 2024-05-11:
### Question

Hello two days ago I installed Windows 11 on my Mipad 5 without any real problem, Then I went to install AtlasOS to improve the perfomance on Windows 11 mode, I installed succesufelly but I noticed that for example when I connected my mouse thorugh type C didn't work, it worked if I restarted the tablet and the device was already connected, and also it didn't work well when carging it was not detecting. Well the thing is I turned off my tablet and I plugged the cable but when I did that the tablet turned on but it started looping in MIUI logo without end, I started the device on fastboot mode and I flashed to UEFI and I could start on Windows 11 again, but I can not find a way to turn my device in to android again, I flashed through fastboot with the normal boot img and also the one patched with magisk and not work at all, so if you could know a way to turn my device again to Android I'll be very happy, Thanks.

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

n00b69, 2024-05-12:
Do you use stock MIUI?

KevsCoder, 2024-05-14:
> Do you use stock MIUI? Try reflashing the stock rom with MiFlash.

I thought about it, I can do it with the installation of Windows 11? Or do I have to remove first Windows (With the tutorial's guide) in order to reflashing the stock?

n00b69, 2024-05-15:
> > Do you use stock MIUI? Try reflashing the stock rom with MiFlash.
> 
> I thought about it, I can do it with the installation of Windows 11? Or do I have to remove first Windows (With the tutorial's guide) in order to reflashing the stock? And where can I get the stock Rom?

https://xiaomifirmwareupdater.com/miui/nabu/

You do not need to remove Windows. Just make sure to NOT lock the bootloader in MiFlash.

closed 2024-06-09 by n00b69