erdilS/Port-Windows-11-Xiaomi-Pad-5 #194: The WINNABU partition can't be found. Please ensure that you have run MSC and haven't missed the device partitioning step

blimer, 2024-05-29:
### Description

I have unlocked bootloader and rooted device (xiaomi pad 5 6/64).
I previously installed Windows using an old guide, and it worked without any issues.
Now i am stuck in step "Assign letters to WINNABU and ESPNABU partitions"
[image]
[image]
[image]

### Reproduction steps

```bash
Fresh install rom V14.0.8.0.TKXEUXM
Root the rom via magisk
Get Stuck in step: Assign letters to WINNABU and ESPNABU partitions
```

### Logs

```bash
Press Y to continue, or N to cancel.
Do you want to proceed? (Y/N): Y
The WINNABU partition can't be found. Please ensure that you have run MSC and haven't missed the device partitioning step

press any key to exit
```

### Agreement

- [X] I am using the latest Drivers and UEFI.
- [X] I am running Windows 11 on the tablet.
- [X] There is no issue already describing my problem.

Misha803, 2024-05-29:
looks like you doesn't partitioned the device

Misha803, 2024-05-29:
oh wait, you reinstalling Windows, right?

Misha803, 2024-05-29:
btw can you contact me on Telegram - https://t.me/misha_803. It will be better to speak there

Misha803, 2024-05-29:
oh wait, you reinstalling Windows maybe ?

closed 2024-06-18 by n00b69