erdilS/Port-Windows-11-Xiaomi-Pad-5 #213: [QUESTION] - Battery life

ImMega, 2024-07-27:
### Question

Hi, I was wondering, how long does the battery last from 100% while idling in Windows? I was thinking of trying this yet again now that even more issues are fixed

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

dikeckaan, 2024-07-29:
> ### Question
> Hi, I was wondering, how long does the battery last from 100% while idling in Windows? I was thinking of trying this yet again now that even more issues are fixed
>
> ### Agreement
> * [x] [...]

Hello,

My SOT is more than 6 hours even my tablet is on hardcore usage. Most cases nearly 10 hours.

Battery backup is good, better than most laptops

ericf2k24, 2024-09-26:
My Mipad 5 on win 11 usually last the whole day, but I've never allowed it to go lower than 20% of battery charge

tvorogo, 2026-02-22:
> [@tvorogo](https://github.com/tvorogo) Release is not release therefore the issue still exists
> 
> Release cannot be a release until someone fucking sends the .fd for dbkp

I did send it, but if you want it for buttons, it won't be there unless Remtrik provides an FD with the fatboot removed. I'm not ready to take responsibility for providing UEFI from self-made resources.

tvorogo, 2026-02-22:
and also in the new release the bug is not fixed

n00b69, 2026-02-22:
@tvorogo If it is not fixed why do you want the issue to be closed