erdilS/Port-Windows-11-Xiaomi-Pad-5 #214: [QUESTION] - Tiny 11 ARM64

japhethLG, 2024-07-29:
### Question

Hi, first of all, I'd like to thank the devs for this awesome project! Is it possible to use Tiny11 instead of the standard Win11? Would there be any issues with the drivers? Here's the link for reference: https://archive.org/details/tiny11a64 

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

n00b69, 2024-08-03:
Possible, but make sure to disable Windows update once you boot into the system. Tiny11 breaks after you try to update.

kujourin, 2024-08-03:
Can I install win server 2025 on my device? It is the same series as win11.

n00b69, 2024-08-04:
If there is an ARM64 version available

closed 2024-08-20 by n00b69