erdilS/Port-Windows-11-Xiaomi-Pad-5 #229: [QUESTION] -  Cannot disbale CABC (Content adaptive brightness control) on Windows 10

RedSTARO, 2024-08-13:
### Question

 Cannot disbale CABC (Content adaptive brightness control) on Windows 10 
I tried version 22h2 21h2, they have same problem and CABC are not appear in old version UEFI (2024-01-15)
There is any way disable it please?

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

RedSTARO, 2024-08-13:
https://github.com/map220v/MU-sm8150pkg/commit/6d9a52b61583397ff44f7c6a165e858ba251fb3d

Commit add this feature here, it can be disbaled in Windows11, but can not be removed from Windwos10

n00b69, 2024-08-14:
UEFI with disabled CABC is available here: https://t.me/nabuwoa/243828

RedSTARO, 2024-08-14:
Thanks, already got files.
Thanks for your great work again!~~~~

closed 2024-08-14 by RedSTARO