erdilS/Port-Windows-11-Xiaomi-Pad-5 #231: [QUESTION] - Install on xiaomi pad 5 cn 8gb ram

2020loryx, 2024-08-14:
### Question

It is possible to install windows on a xiaomi pad 5 china version with 8gb ram?

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

n00b69, 2024-08-14:
Yes it is possible, the only difference is RAM.

The only exception is if you are using a Chinese version with modded UFS (512GB/1TB), these are currently not supported by the partitioning script (because we don't have some values needed to make it work).

2020loryx, 2024-08-15:
Ok thak you, so i don't need other drivers and windows will recognize the whole 8gb right?

n00b69, 2024-08-16:
Correct

2020loryx, 2024-08-16:
Thank you for the help

closed 2024-08-16 by 2020loryx

Gezine, 2024-10-19:
Sorry to revive this issue but where can I get information about 8GB mi pad5?
Never heard of it.

n00b69, 2024-10-19:
It is only available in China

Gezine, 2024-10-19:
So we can use same uefi file for 8GB version?