erdilS/Port-Windows-11-Xiaomi-Pad-5 #248: [QUESTION] - VS CODE + NODE

Danxguardian, 2024-09-29:
### Question

### Question
Hi everyone Im a developer and Im looking for a portable way (other than a laptop, mac or surface) to be able to work on personal projects in cafes or while waiting for someone (regularly nothing complicated frontend mostly)

I came here because of a YT video and I had not thought about this option, until now I only thought about Android + Termux

Well my questions

1. Does VS Code (Node) support?
2. Does it support terminals to run a local server?
3. Does it support something heavier like Blender, Android Studio (not for emulating devices)?

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

Oery, 2024-10-08:
Hey, VS Code works perfectly fine, Node too. 
Blender ARM64 builds are not compatible. I haven't tried the emulated x64 version, but even if it could start, it would probably not run well.  Android Studio doesn't run very well and become nearly unusable during project setup / builds.
Honestly I could see myself use it to work on VS Code but other than that, not really.

n00b69, 2024-10-20:
Intellij runs very well as a replacement for Android Studio, it has a native ARM64 build.