erdilS/Port-Windows-11-Xiaomi-Pad-5 #249: [BUG] - <auto-rotation unavailable>

KOTORIcc, 2024-09-30:
### Description

After updating the latest drivers , the screen auto-rotation feature seems to be unavailable
[image]

### Reproduction steps

```bash
osversion: windows 11 23h2
```

### Agreement

- [X] I am using the latest Drivers and UEFI.
- [X] I am running Windows 11 on the tablet.
- [X] There is no issue already describing my problem.

n00b69, 2024-10-02:
Show me the sensor device(s) in device manager

timoxa0, 2024-10-07:
Also check if HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\PriorityControl\ConvertibleSlateMode is set to 1 in the registry, if so reset it to 0 to activate autorotation

KOTORIcc, 2024-10-08:
it works，thanks.👌

closed 2024-10-17 by Misha803