erdilS/Port-Windows-11-Xiaomi-Pad-5 #256: How to charge the tablet after shut down?

xboltzx, 2024-10-24:
### Question

While on Win 11 mode, I select shut down with the charger plug in. It shut down but after awhile it boots up to win 11 again.
If I shut down without the charger it will not boot up, which is great! But the moment i plug in the charger without pressing the power button it will auto boot to Win 11.

It it normal? What i am trying to do is to shut down the tablet and charge it.

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

n00b69, 2024-10-26:
Not possible, the only way to do so is to add -s to a shortcut of C:\ProgramData\sta\sta.exe.

This will flash boot.img and shutdown. Offline charging works with boot.img flashed.

xboltzx, 2024-10-29:
Thanks for the reply, the mentioned file path of C:\ProgramData\sta\sta.exe is not there.
I can only reach to C:\ProgramData there is no sta folder. Is there any thing wrong with my tablet?

n00b69, 2024-10-29:
No, it just means you installed Windows before the sta path was changed.

Create the sta folder, and put this exe inside.

https://github.com/n00b69/woa-helper/blob/dev/app/src/main/assets/sta.exe

xboltzx, 2024-10-31:
GREAT!!! this is what i had wanted. thanks alot for all the help!

closed 2024-10-31 by xboltzx