erdilS/Port-Windows-11-Xiaomi-Pad-5 #259: [QUESTION] - <title>Questions about Device Type Identification

RealShiinaMashiro, 2024-10-30:
### Question

This message has been translated by an Online Translator, and we apologize for any misunderstanding that it might cause.
The new driver causes Microsoft to recognize the device as a phone instead of a Pad. Is there a way to change the device type back to a Pad or Notebook?

以下英语内容是通过机器翻译得到的，如有错误，深感抱歉。
更新了新的驱动后，会让微软把设备识别成手机，而不是平板。有什么方法，可以把设备类型恢复成平板或者电脑吗？
[image]

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

n00b69, 2024-11-03:
In 23H3 it was recognized as a tablet?

Also usinf 24H2 is not recommended. It can corrupt your GPT.

n00b69, 2024-12-17:
This was actually my fault, however it should be fixed in the latest version of drivers, should you wish to reinstall

closed 2024-12-17 by n00b69