erdilS/Port-Windows-11-Xiaomi-Pad-5 #282: [BUG] - Another case of white lines after sleep.

MinePossu, 2024-11-11:
### Description

I'm not sure if issue should be tracked. But since this issue is known il just report a another case of it happening.

Maybe others can report their unlucky nabus on this issue.

This nabu bought from ebay (italy) used.

### Reproduction steps

```bash
Sleep in windows and then dread the pain.
```

### Agreement

- [X] I am using the latest Drivers and UEFI.
- [X] I am running Windows 11 on the tablet.
- [X] There is no issue already describing my problem.

n00b69, 2024-11-21:
Rare issue affecting some panel variants, sadly cannot be fixed, and it is also impossible to know if it will affect your panel or not before actually installing Windows.

closed 2024-11-21 by n00b69

MinePossu, 2024-11-21:
Affects my panel.

Hakzel, 2025-04-24:
Also affects my panel. So I don't think it is too rare of a problem

I bought my pad new directly from XIaomi and it never has been taken apart. So it is still completely at factory state

Not being able to use sleep is the only downside of this otherwise great project which I really love!