erdilS/Port-Windows-11-Xiaomi-Pad-5 #285: [QUESTION] - Partition Custom ROM Xiaomi Pad 5

smartfrenard, 2024-11-21:
### Question

I have custom ufs, upgrade from 256 Gb to 512 Gb for Xiomi Pad 5. How to partition for custom rom 512 Gb?
[image]
[image]

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

n00b69, 2024-11-21:
Please run the below command and send a picture

```cmd
adb shell blockdev --getsize64 /dev/block/sda
```

After that, use these steps to partition:

```cmd
adb shell sgdisk --resize-table=128 /dev/block/sda
```

```cmd
adb shell parted /dev/block/sda
```

```cmd
rm 31
```

Replace 256GB with whatever you want Android to have
```cmd
mkpart userdata ext4 10.9gb 256gb
```

Replace 256 with the value you used in userdata
```cmd
mkpart esp fat32 256gb 256.35gb
```

Same here with 256
```cmd
mkpart win ntfs 256.35gb 509gb
```

```cmd
set 32 esp on
```

```cmd
quit
```

```cmd
adb shell mkfs.ntfs -f /dev/block/sda33 -L WINNABU
```

```cmd
adb shell mkfs.fat -F32 -s1 /dev/block/sda32 -n ESPNABU
```

Format data in ofox and check if android boots, then reboot back to recovery and do the steps on the second page

smartfrenard, 2024-11-23:
adb shell blockdev --getsize64 /dev/block/sda

[image]

smartfrenard, 2024-11-24:
Next Step page 3 guide/English/3-install-en.md, can't run execute msc?

[image]
[image]

n00b69, 2024-11-24:
Give me some time, I'll add 512GB support to the recovery

n00b69, 2024-11-24:
Boot this recovery

https://github.com/erdilS/Port-Windows-11-Xiaomi-Pad-5/releases/download/1.0/512.img

Try `adb shell partition $`

Then try `adb shell msc`

smartfrenard, 2024-11-24:
Sorry, I forgot to tell you my custom nabu 8/512Gb.
Successfully install driver using latest version but always restart and blue screen? is the driver incompatible for nabu custom 8/512Gb?

[image]
[image]
[image]
[image]
[image]
[image]
[image]

https://github.com/user-attachments/assets/4c6ff93e-5dbb-430c-b9e5-a855d090dc3e

n00b69, 2024-11-24:
Try `adb shell format` and try reinstall Windows, might be just a random bug.

n00b69, 2024-12-21:
Closing, assuming it was fixed

closed 2024-12-21 by n00b69