erdilS/Port-Windows-11-Xiaomi-Pad-5 #286: [QUESTION] - Pad 5 Pro support

KuDeSnik33ra, 2024-11-23:
### Question

Hello! Thanks for your work! Can you, please, clarify what devices are supported by your guides? Is only Pad 5 supported, or Pad 5 Pro too? Isn't there differences on year of release? Are the 6/128 and 8/256 pads both supported? 

### Agreement

- [X] I am asking a sensible question.
- [X] I am not trying to waste time.
- [X] There is no issue already describing my problem.

n00b69, 2024-11-24:
Not supported and will never be. Incompatible chip.

closed 2024-11-24 by n00b69

KuDeSnik33ra, 2024-11-24:
Thanks for the answer, what about other questions? 8ram/256ssd and 6ram/128ssd works both? Year of release does not matter?

n00b69, 2024-11-24:
Any Xiaomi Pad 5 with Snapdragon 860 works.