erdilS/Port-Windows-11-Xiaomi-Pad-5 #348: [QUESTION] - Where is the source code of the nt36523 driver used by nabu?

Aquarius223, 2025-05-08:
### Question

I only found the compiled product [(nt36523-driver)](https://github.com/Project-Aloha/windows_oem_xiaomi_nabu/tree/master/Drivers/Touch)

### Agreement

- [x] I am asking a sensible question.
- [x] I am not trying to waste time.
- [x] There is no issue already describing my problem.

remtrik, 2025-05-08:
@Aquarius223 Just out of curiosity, why do you want it?

Aquarius223, 2025-05-08:
> [@Aquarius223](https://github.com/Aquarius223) Ah I've seen that commit, it's already in windows driver. As it turns out I just forgot to release an update for it. Will be included in the next driver update

Oh, by the way, due to the new firmware, xiaomi has updated some magic changes about the pen. If you use the new firmware, it may affect the use of some third-party pens, such as being unable to write and click events not working, so we should create a node similar to Linux for users to switch firmware

Aquarius223, 2025-05-08:
> [@Aquarius223](https://github.com/Aquarius223) That's not really possible when talking about windows

OK, I will close this issue.

closed 2025-05-08 by Aquarius223